micro-manager/mmCoreAndDevices #502: Arduino realize 4 analogue output with TLV56X8

asincerity, 2024-10-11:
Hello, I noticed that arduin can be combined with TLV5618 to achieve two channels of analog output, but this is usually not enough. I found a TLV56X8 DAC module which supports outputting 4 analog outputs. Can I ask if this firmware can be supported with certain modifications to work with this module. Below is the circuit diagram of the TLV56X8, if this is possible this will fully utilize the capabilities of the arduino and be easy to use for researchers, thanks for being able to read this question. 

[image]

asincerity, 2024-10-14:
The channels of the four-channel DAC output signals are realized by CS1 and CS2, while in the recommended TLV5618, there is only one CS channel, and the channels are switched by the high and low voltages of this pin. So it seems that we only need to add in the Arduino script the switching of the channels by the level of CS1 and CS2, the original CS channel is connected to pin 5 of the Arduino, and now with four channels we can add CS2 to pin 6. Then we should need to add two DAC channels in the Arduino driver of the micromanager, and I can see that it can be easily done in the Arduino driver. I see in micromanager there is a dll file for arduino which should be the driver, but I don't have the knowledge and experience, is it easy to accomplish this?

Translated with DeepL.com (free version)

nicost, 2024-10-15:
That looks like a very nice board. Regretfully, I do not see it for sale in the US, and also can not find more documentation about it.  It does look like you need at least one extra digital pin to control that board.  This will need some modifications to the Arduino firmware code and likely also (modest) changes to the Micro-Manager Arduino device adapter.  Let us know how you progress!

asincerity, 2024-10-16:
Yes, this requires an additional pin6 to control CS2. CS1 and CS2 to switch between the four channels. In fact, even with the same wiring as the recommended TLV5618 and only pin5 connected to CS1, the board's OUT1 and OUT2 analog outputs can be controlled by DAC1 and DAC2 in the micromanager. If pin5 is connected to CS2, it can realize the control of CS2. This means that we only need to add a pin6 control condition for channel selection when outputting analog signals in Arduino. At the same time, we need to add the corresponding DAC2 DAC3 in the Arduino device adapter. But I don't have the relevant knowledge, so I would like to ask the author of this firmware and adapter or other seniors. I would like to ask if you can implement these changes quickly or have some guidance?
[image]

nicost, 2024-10-28:
I am afraid that only people with access to that board can do this.  Where did you get it from?

asincerity, 2024-10-29:
Thank you for your reply. I got it from Taobao shopping platform in China, this is a screenshot of the store. I wonder if you can find it? In fact, the original situation was to use the high and low levels of pin5 to switch two channels, now you need to use the combination of high and low levels of pin5 and pin6 to switch 4 channels.
[image]

asincerity, 2024-11-20:
Thank you for your help. I solved the problem by adding two DAC3 and DAC4 to the arduino adapter in the downloaded source code. basically just add the positions of the corresponding DAC1 and DAC2. Then follow the instructions in “https://micro-manager.org/Building_MM_on_Windows”to compile the dll files and replace them in the corresponding version of micromanager. I don't know why, but the dll file I compiled is much bigger than the original one. Then I added the channel response to the arduino script, and now I can use the arduino and the 4-channel DAC module in micromanager.Even an 8-channel DAC module can be added later.
[image]

closed 2025-02-03 by nicost