micro-manager/mmCoreAndDevices #512: Arduino: make number of patterns in sequence dynamic

nicost, 2024-10-22:
This number is now hard coded in the device adapter.  Since there are now plenty of Arduinos with more memory than there used to be, it would be nice to have the device adapter query this number on startup.

simonecoppola, 2024-10-23:
Not a long term solution, but I have recompiled the device adapter and changed the firmware to support a max sequence length of 256 (which works on the latest Arduino UNO Rev3). I've put it [here](https://github.com/simonecoppola/arduino-micromanager) in case somebody else needs it in a pinch.

nicost, 2024-10-29:
I just updated the firmware to version 3. That version will report how many sequence patterns it can accept. I also updated the Device adapter to dynamically read that number. The default is now 256, which appears to work well. I advise going much higher in an UNO as it has limited RAM. Should be in tonight's build and later.

closed 2024-10-29 by nicost