micro-manager/mmCoreAndDevices #522: Missing SLN and .props in branch device-interface-70

TIS-Stefan, 2024-11-01:
Hello 

I  build a camera device adapter. The adapter compiles fine with files from main branch. But the downloadable version on https://micro-manager.org/ is version 2.0.0, which loads device dlls with DEVICE_INTERFACE_VERSION=70. Thus, my DLL will not be loaded. 
Faking 71 to 70 in MMDevice.h lets the DLL being loaded and cameras being listed, but adding a camera will lead to an unexpected program end. 
Error message in Visual Studio is
```
error MSB4019: The imported project "micro-manager\mmCoreAndDevices\buildscripts\VisualStudio\MMCommon.props" was not found. Confirm that the expression in the Import declaration "..\buildscripts\VisualStudio\MMCommon.props" is correct, and that the file exists on disk.
```
I fail to get a solution file for Visual Studio when using the device-interface-70 branch. Question: How do I get that? 

Thank you in advance!

Stefan

nicost, 2024-11-11:
Please everyone, use the latest nightly build rather than the 2.0 build.  Way more features and much more stable.  Unless there is some specific reason to stick to a very old version (there is some hardware that only works with older versions) there really is no good reason no to.

With DEVICE_INTERFACE_VERSION=70 we had not yet completed migrating all the hardware building code into a single unit.  The micromanager.sln file is in a corresponding commit of the micromanager repository.  You can roughly find a commit close to the release date, clone that, then move your mmCoreAndDevices branch in there.

TIS-Stefan, 2024-11-11:
Hello 

I would like to, but it is hard enough to ask the users to change from version 1.4 to 2.0.0. Also most users are not used to Github and thus prefer the download from the official website.

Stefan

closed 2024-11-11 by TIS-Stefan

dpshepherd, 2024-11-12:
The nightly builds are distributed as executables:

[Windows](https://download.micro-manager.org/nightly/2.0/Windows/)
[Mac](https://download.micro-manager.org/nightly/2.0/Mac/)

TIS-Stefan, 2024-11-12:
Hello 

Thank you very much!

Stefan