micro-manager/mmCoreAndDevices #541: Visual Studio 2022

marktsuchida, 2024-12-19:
Updating the projects is likely trivial; most of the work will be in updating the build worker VM images.

nicost, 2024-12-20:
Bumping this after discovering that VS2019 can no longer be downloaded (without an account paying $$ to MS).

marktsuchida, 2025-02-08:
See also: https://github.com/micro-manager/micro-manager/pull/2062

marktsuchida, 2025-02-12:
The nightly build has now been switched to a new Windows image containing VS2022 (with VS2019 build tools). (It only took 2 and a half days to get that working!) This has been tested and will apply starting with the next nightly build.

This means that we are now almost ready to switch the projects to use the VS2022 toolset, but I suggest that we wait until a time when we don't have any major pull requests pending (or at least none that add, remove, or rename projects), because touching all the project files can complicate merging. We'll also need to update the VC redistributable that we ship with our installer.

marktsuchida, 2025-02-12:
(I don't think the VS2019 build tools installable with VS2022 are going away any time soon; VS2019 EOL is still a few years ahead.)

marktsuchida, 2025-02-26:
All device adapters build, except for the 3 for which I have created sub-issues.

marktsuchida, 2025-05-06:
- [ ] Confirm that the vcredist we package with the Windows installer is at least as as new as the VS2022 we use to build.

commit e7bda778 (2025-05-20): Switch to VS2022 toolset (v143)

Closes #541.
closed 2025-05-20 by marktsuchida