micro-manager/mmCoreAndDevices #551: IDS Peak Hub - Specified property limits are not valid

Henttinen, 2025-01-20:
Hello @Lars-Kool,

I am running **MMStudio Version 2.0.3 20250114** and currently trying to connect a IDS camera, model [U3-3560XCP-C-HQ Rev.1.2](https://en.ids-imaging.com/store/u3-3560xcp-rev-1-2.html) and I am getting the specified error when trying to add the camera.

Here is a screenshot of the error when I try to add the camera.

[image]

Here are the initialization properties I see for the IDS Peak Hub and the IDSCamera0

[image]

[image]

I have tested that the camera works through the IDS Peak Cockpit and used it to capture images after MMStudio didn't work.

Lars-Kool, 2025-01-20:
Hi @Henttinen , I've been made aware that the XCP series is a bit different than the other IDS cameras. I've put a temporary fix here: https://forum.image.sc/t/anyone-working-on-an-ids-peak-device-adapter/62990/52?u=lars-kool. I will add my fix to the IDS device adapter asap.

Best,
Lars

Lars-Kool, 2025-04-02:
Hi @Henttinen,

Does your camera work with the new nightly build of MM? If not, could you provide the CoreLog?

Best,
Lars

Lars-Kool, 2025-04-03:
Hi @Henttinen,

Ahh I see, your camera only supports very specific data layouts, none of which I was using to check whether the camera supports RGB. I'm currently reworking the device adapter to support more features, and I will fix this bug as well. The device adapter below should fix the issue in the mean time.

[mmgr_dal_IDSPeak.zip](https://github.com/user-attachments/files/19583776/mmgr_dal_IDSPeak.zip)

Best,
Lars

Lars-Kool, 2025-11-27:
The new IDS device adapter is out, and this issue should be fixed. @marktsuchida This issue can be closed.