micro-manager/mmCoreAndDevices #578: DahengGalaxy doesn't build with VS2022 (v143) toolset

marktsuchida, 2025-02-26:
When the platform toolset is upgraded to v143 (testing for a future upgrade to VS2022), we get the following error.
Presumably the `volatile` is a mistake, but it's part of the vendor SDK, so ideally we'd get an updated version from them.

```text
------ Build started: Project: DahengGalaxy, Configuration: Release x64 ------
ClassGalaxy.cpp
C:\Users\mark\devel\3rdpartypublic\DahengGalaxy\inc\GXSmartPtr.h(38,2): error C7731: 'volatile' is not allowed on a constructor declaration
(compiling source file 'ClassGalaxy.cpp')
Done building project "DahengGalaxy.vcxproj" -- FAILED.
```

nicost, 2025-02-27:
The SDK we have in 3rdpartypublic is the latest available from their website.  

There also is an SDK in C.  Preferably we would use that (as it would also work on linux, where they apparently do not have a C++ version), but that means re-writing the device adapter code.

marktsuchida, 2025-02-27:
For the purpose of building with VS2022 tools, we could patch the SDK header (just remove the word `volatile`; it is not doing anything anyway). I just meant we should let them know of the issue.

commit bc36714c (2025-05-20): DahengGalaxy: Upgrade SDK to latest

This allows building with VS2022/v143 toolset.

Closes #578.
closed 2025-05-20 by marktsuchida