micro-manager/mmCoreAndDevices #620: ASI LX-4000 timeout during timelapse

dpshepherd, 2025-04-17:
We are facing an issue with a multi-channel, z stack, timelapse experiment. When we reduce the exposure time in the MDA window, we eventually get a `TERM_TIMEOUT error occured`. If we lengthen the exposure time, this problem goes away.

I've attached a problem report log. The error shows up at timestamp `2025-04-16T16:39:44.759802`.

Any suggestions on how to get around this issue? We are using the old acquisition engine, because the new one was (is) resetting the Z stage to the bottom of a stack instead of the middle after every z-stack.

This is on the newest nightly build. Any suggestions on how to solve the issue?

Thanks!

[20250416_Problem Report.txt](https://github.com/user-attachments/files/19786522/20250416_Problem.Report.txt)

nicost, 2025-04-18:
It looks like the StageControl window's "Polling updates" is checked.  Try unchecking it before running the MDA.  

Background: the ASI XYStage and ASI Filterwheel are both connected through the same serial port (COM16).  These two devices are in different device adapters.  When one thread is talking to the FilterWheel (in this case, the main acquisition thread) and the other thread (the polling updates from the StageControl window) is talking to the XYStage, they can talk more or less at the same time, and mix-ups will happen. To avoid these kind of things, there is locking at the level of the device adapter.  However, since these two devices are in different device adapters, that locking mechanism does not help.  

I will look at the new acquisition engine and z stack settings (I may open a new issue for that).

nicost, 2025-04-18:
I fixed one obvious problem with the new acquisition engine (https://github.com/micro-manager/micro-manager/pull/2092).  If there are more, please let me know by posting a problem report and describing the issue.

dpshepherd, 2025-04-18:
Thanks Nico! I'll ask group to try and report back.

dpshepherd, 2025-04-24:
Unchecking the "polling updates" does fix the issue. thanks!