micro-manager/mmCoreAndDevices #652: ImageJ's JVM crashed when opening without connecting the Balser camera

zhaolun7, 2025-05-23:
Hi developers,

Recently I need to change the camera to Balser. Following the instruction(https://micro-manager.org/BaslerPylon), I install the Basler driver with version 7.1. It works with Micro-Manager and my application.
I find if I don't connect the USB of Basler camera to the computer, my application's JVM will crash directly when trying to connect to camera. This took me some days to debug. Today I test it with Micro-Manager, ImageJ crashed too.

I barely know C++ so I guess when I don't have a camera connected, a C++ error is not passed to Java, causing the crash. I tried many nightly build versions and they both have this issue. 

I really appreciate your help and suggestions.

[image]

[image]

zhaolun7, 2025-05-23:
Another problem is this [camera](https://www.edmundoptics.com/p/basler-ace2-a2a4504-18umbas-monochrome-usb3-basic-camera/44043/?gad_source=1&gad_campaignid=21420803257&gbraid=0AAAAADn0vHZd2R58FR00zDlpoJSIyyEaJ&gclid=Cj0KCQjwlrvBBhDnARIsAHEQgOR-b3Y06cDN9L1MOHOua-HNiz7f6k14Vmdm-4nf9RiMM2grORzgQnIaAj5uEALw_wcB) only support 12 bit pixel depth, is there a way to specify MSB or LSB alignment?

marktsuchida, 2025-05-23:
Thanks, I think the crash is a simple mistake and I have a possible fix.

Regarding msb/lsb alignment, Micro-Manager always aligns values of less than 16-bit depth to the lsb, and can usually query the camera for the actual bit depth. If you are seeing values outside the range of 0-4095, this might be a bug in the BaslerPylon device adapter, so let us know (please create a separate issue).

(In the Script Panel, `mmc.getImageBitDepth();` should return 12; if you get an error, try snapping an image first.)

If you want to msb-align for the purpose of downstream processing, you will need to multiply the pixels by 16 as part of that processing. But if you just want to see the image with good contrast, the histogram controls in Micro-Manager should allow this.

closed 2025-05-23 by marktsuchida

zhaolun7, 2025-05-23:
@marktsuchida Yes it directly crashed and generated a `hs_err_*.log`: 

[image]

[hs_err_pid20684.log](https://github.com/user-attachments/files/20418709/hs_err_pid20684.log)

[image]

marktsuchida, 2025-05-23:
That looks like the error I saw when we upgraded our Windows C++ compilers recently. The problem is that the Java runtime contained an older Microsoft C++ runtime that was no longer compatible (https://github.com/micro-manager/micro-manager/pull/2101).

Try upgrading your JDK to the latest JDK 11 (e.g. Temurin or Zulu).

zhaolun7, 2025-05-23:
Thanks. I will wait for the nightly build and test them.

PR #653 (2025-05-23): Basler: Don't throw exception when no camera
Instead, leave the SerialNumber choices empty and let Initialize() return the error.

Closes #652.