micro-manager/mmCoreAndDevices #707: Large overhead for hardware sequencing when using PVCAM cameras in multicamera adapter

ieivanov, 2025-08-07:
I've run into a a bit of an esoteric bug which I've managed to trace down to the PVCAM device adapter. I'll try to summarize the findings I've gathered so far, but please ask any questions that can help clarify this story.

In our experiment we would like to acquire Z stacks over many positions. We have the microscope set up such that 2-3 channels can be acquired at the same time using the multicamera device adapter. We would also like to use hardware sequencing to move through Z slices before moving onto the next XY position. My current observation is that enabling hardware sequencing leads to large overhead at every XY position which makes the acquisition overall 2-3x slower (when the intention was to speed it up through HW sequencing over Z slices).

I've managed to reproduce this issue with only demo devices plus one PVCAM camera - in this case a Prime BSI Express (I've seen the same behavior with a Prime BSI camera as well). For purposes of this test I've simulate moving between XY positions with a timelapse with 0.1 second interval between timepoints, i.e. hardware sequences are restarted at each timepoint. Take a look at this config file [config file](https://github.com/user-attachments/files/21674513/MMConfig_demo_PVCAM_MultiCam.txt) and this beanshell script:

```java
// Prepare hardware
mmc.setExposure("DCam", 100);
mmc.setExposure("DCam2", 100);
mmc.setExposure("Prime BSI Express", 100);
mmc.setProperty("Prime BSI Express", "ReadoutRate", "100MHz 16bit");
mmc.setProperty("DCam", "OnCameraCCDXSize", 2048);
mmc.setProperty("DCam", "OnCameraCCDYSize", 2048);
mmc.setProperty("DCam", "PixelType", "16bit");
mmc.setProperty("DCam2", "OnCameraCCDXSize", 2048);
mmc.setProperty("DCam2", "OnCameraCCDYSize", 2048);
mmc.setProperty("DCam2", "PixelType", "16bit");
mmc.setProperty("Core", "Focus", "Z");

// Prepare acq settings
acq_manager = mm.getAcquisitionManager();
settings_builder = acq_manager.sequenceSettingsBuilder();
settings_builder.useFrames(true);
settings_builder.numFrames(10);
settings_builder.intervalMs(100);
settings_builder.useSlices(true);
settings_builder.sliceZBottomUm(0);
settings_builder.sliceZTopUm(20);
settings_builder.sliceZStepUm(1);
settings_builder.keepShutterOpenSlices(true);
settings_builder.root("D:\\testing");  // Modify acquisition directory as needed
settings_builder.prefix("speed_test");
settings_builder.save(true);
settings = settings_builder.build();

// Run non-sequenced acquisition with Prime BSI Express camera
mmc.setProperty("Core", "Camera", "Prime BSI Express");
mmc.setProperty("Z", "UseSequences", "No");

long startTime = System.currentTimeMillis();
acq_manager.runAcquisitionWithSettings(settings, true);
long endTime = System.currentTimeMillis();
long elapsedTime = endTime - startTime;

// Establish baseline
print("Elapsed time: " + elapsedTime/1000 + " s");  // 32 seconds

// Run sequenced acquisition with Prime BSI Express camera
mmc.setProperty("Core", "Camera", "Prime BSI Express");
mmc.setProperty("Z", "UseSequences", "Yes");

long startTime = System.currentTimeMillis();
acq_manager.runAcquisitionWithSettings(settings, true);
long endTime = System.currentTimeMillis();
long elapsedTime = endTime - startTime;

// Acquisition is faster, as we'd expect
print("Elapsed time: " + elapsedTime/1000 + " s"); // 24 seconds

// Run sequenced acquisition with Prime BSI Express and demo camera
// in multicamera adapter
mmc.setProperty("Multi Camera", "Physical Camera 1", "Prime BSI Express");
mmc.setProperty("Multi Camera", "Physical Camera 2", "DCam");
mmc.setProperty("Core", "Camera", "Multi Camera");
mmc.setProperty("Z", "UseSequences", "Yes");

long startTime = System.currentTimeMillis();
acq_manager.runAcquisitionWithSettings(settings, true);
long endTime = System.currentTimeMillis();
long elapsedTime = endTime - startTime;

// Acquisition is 2x slower, which is a bug
// Between timepoints the circular buffer is resized several times
print("Elapsed time: " + elapsedTime/1000 + " s"); // 49 seconds

// Run sequenced acquisition with only one Prime BSI Express camera
// in multicamera adapter
mmc.setProperty("Multi Camera", "Physical Camera 1", "Prime BSI Express");
mmc.setProperty("Multi Camera", "Physical Camera 2", "Undefined");
mmc.setProperty("Core", "Camera", "Multi Camera");
mmc.setProperty("Z", "UseSequences", "Yes");

long startTime = System.currentTimeMillis();
acq_manager.runAcquisitionWithSettings(settings, true);
long endTime = System.currentTimeMillis();
long elapsedTime = endTime - startTime;

// Acquisition is as fast as acquiring data directly with the Prime BSI Express
// i.e. having the Prime BSI Express camera in the multicamera adapter
// is not the source of the problem
print("Elapsed time: " + elapsedTime/1000 + " s"); // 24 seconds

// Run sequenced acquisition with two demo cameras in multicamera adapter
mmc.setProperty("Multi Camera", "Physical Camera 1", "DCam");
mmc.setProperty("Multi Camera", "Physical Camera 2", "DCam2");
mmc.setProperty("Core", "Camera", "Multi Camera");
mmc.setProperty("Z", "UseSequences", "Yes");

long startTime = System.currentTimeMillis();
acq_manager.runAcquisitionWithSettings(settings, true);
long endTime = System.currentTimeMillis();
long elapsedTime = endTime - startTime;

// Acquisition is fast. The sequence buffer does not get resized between timepoints
print("Elapsed time: " + elapsedTime/1000 + " s"); // 23 seconds
```

The issue seems to be specific to having a PVCAM camera plus another camera in the multicamera adapter and seems to be linked to unnecessary resizing of the circular butter. I've seen the same problem with Prime BSI Express + Andor Zyla cameras in the multicamera adapter and Prime BSI Express + Prime BSI cameras in the multicamera adapter. The problem is not present when using two demo cameras or two Zyla cameras in the multicamera adapter.

@tomhanak @nicost could you help me resolve this issue? Nico, we'd be happy to lend you a Prime BSI Express camera if you have bandwidth to take a look.

marktsuchida, 2025-08-08:
Thanks for the detailed script!

How are you finding out that the circular buffer is being resized? I'm asking in part because the PVCAM adapter indeed has code (called via `startSequenceAcquisition()`) that resizes buffers, but it refers to both the MMCore "circular" buffer and the PVCAM driver's circular buffer, and it's not clear which (if not both) is at issue here.

ieivanov, 2025-08-08:
Good question! I've been monitoring the usage and size of the image buffer that's shown by the GUI in Plugins > Developer Tools > Sequence buffer monitor. I see that before every Sequence start it is first reduced in size - I think to 1/C where C is the number of channels in the multicamera adapter, and then returned to its original size which is used during the sequence acquisition, i.e. the buffer resize is unnecessary, and possibly the source of the slowdown I'm experiencing. The buffer usage itself is very low.

ieivanov, 2025-08-14:
Thanks for looking into this @marktsuchida! I really appreciate your help here.

marktsuchida, 2025-09-17:
@ieivanov I believe this should be fixed in the 2025-09-17 nightly, but do let me know if you find otherwise.

ieivanov, 2025-09-18:
Fantastic, thanks @marktsuchida! I'll test it and let you know if I run into any issues.